ARM-software/vscode-cmsis-csolution #173: Create New Solution for device `ARMCM3` does not add required `- pack: ARM::Cortex_DFP`

JonatanAntoni, 2025-01-17:
Type: Bug Report

**Describe the bug**
- OS and Version: MacOS 15
- VS Code Version: 1.96.2
- Extension Version: 1.46.0
- Target Device: ARMCM3 from ARM::Cortex_DFP

Creating a new blank solution via default template for device `ARMCM3` the generated `Blank_solution.csolution.yml` misses the required `- pack: ARM::Cortex_DFP` entry. This results in an error at `cbuild setup`:
```
Execute: cbuild setup Blank_solution.csolution.yml --context-set --packs --update-rte
error csolution: specified device 'ARMCM3' was not found among the installed packs.
use 'cpackget' utility to install software packs.
error csolution: processing context 'Project.Debug+ARMCM3' failed
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to CMSIS pane in side bar
2. Click on 'Create a New Solution'
3. Select Device:
   ARM > ARM Cortex M3 > ARMCM3 
4. Select Project:
   Templates > Black solution
5. Click Create button
6. See error in build console output

**Expected behavior**

The pack the selected device is defined in shall always be added to the generated csolution file.

**Code sample and logs**
- Code sample
- `tasks.json`

**Screenshots**

**Additional context**

ReinhardKeil, 2025-01-18:
I believe this comes from the fact that the IDE still believes it is defined in the CMSIS base pack.  The dialog shows it comes from Arm::CMSIS@5.9.0

[image]

JonatanAntoni, 2025-01-21:
Bug in Core-Tools which needs to be updated.

closed 2025-01-23 by JonatanAntoni