ARM-software/vscode-cmsis-csolution #174: Make uv2csolution errors more visible

ReinhardKeil, 2025-01-18:
Type: Feature Request

For local packs, all uvision examples are shown, even those configured for AC5.

[image]

However this examples cannot be converted.  Unfortunately there is not error message.

Would it be possible to bring these errors to the attention of the user.

```
0 out of 1 target(s) converted
Error: 1 error(s), 0 warning(s):
Errors:
	The target "Target 1" is configured to use Arm Compiler 5.
	CMSIS csolution projects support Arm Compiler 6 only.
	For details on how to manually migrate to AC6, see
	https://developer.arm.com/documentation/100068/0621/Migrating-from-Arm-Compiler-5-to-Arm-Compiler-for-Embedded-6
```

jkrech, 2025-01-21:
We could open a pop-up window in case of error with an option to bring forward the "output" channel for the "uVision to Csolution Converter" (note that the name is too long to be fully displayed in the drop down which is a side effect of changing the title for the command). 
Alternatively we could add some color coded status to the status bar, but I am afraid to clutter the status bar. Adding it to the context set

ReinhardKeil, 2025-01-21:
My suggestion: A pop-up and bringing the output window in front

jkrech, 2025-01-23:
see: https://github.com/ARM-software/vscode-cmsis-csolution/issues/175