ARM-software/vscode-cmsis-csolution #175: uv2csolution converter output

ReinhardKeil, 2025-01-21:
[image]

JonatanAntoni, 2025-01-23:
I cannot reproduce this issue. This is my test:

- Open Visual Studio Code
- Navigate to CMSIS Solution pane
- Click button "Convert a uVision Project to CMSIS Solution"
- Use file picker to choose `test.uvprojx` from provided test project
- Dialog "Project conversion from µVision to csolution completed with warnings. Would you like to open the solution anyway?" pops up.

In the background one can see the conversion output channel to be visible and shows the conversion warnings. Now, one can

- Click button "close" and inspect the warnings.
- Click button "open" and a new instance of VS Code is launched with the conversion result as the active workspace. At that point in time the conversion output is gone and cannot be shown anymore.

ReinhardKeil, 2025-01-23:
My flow was different:

I opened the folder containing the uvproj file, used the Explorer view with right click to this file and used "Convert uVision project".

JonatanAntoni, 2025-01-23:
Thanks for clarifying.

In this case the conversion is expected to set the focus to the "uVision to Csolution conversion" output channel. On my machine, even after removing all csolution files from previous tries and reloading the code window, the conversion output channel is kept in focus.

But, I can imagine that the creation of a new `.csolution.yml` file in the current workspaces triggers `cbuild` right after the conversion step. Hence, the focus is directly switched to the CMSIS Solution output channel. In this case, merging both channels into a common one as done in Arm-Debug/vscode-cmsis-csolution#2073 should do the job.

ReinhardKeil, 2025-01-29:
The issue appears when you open an folder that just contains a *.uvprojx file, right click on the file to start the converter, but not wait until the extensions are activated.

In this case no output is shown

JonatanAntoni, 2025-01-29:
This seems to be a race condition happening during launch/activation process. We need to check if we can postpone the actual conversion until activation is finished.

closed 2025-02-04 by ReinhardKeil

jkrech, 2025-02-17:
CMSIS Solution Extension Version: 1.48.1-14-g1be8064 (internal build from main branch)
All messages from `uv2csolution` now share the output channel "CMSIS Solution"