ARM-software/vscode-cmsis-csolution #176: Unable to change active context in multi-target and multi-project -solution

KimmoVaisanen-AlifSemi, 2025-01-23:
Type: Bug Report

**Describe the bug**
- OS and Version: Windows 11 or Ubuntu 22.04
- VS Code Version: 1.96.4
- Extension Version: 1.46.0
- Target Device: Alif Ensemble E7

We have a project (https://github.com/alifsemi/alif_mlek-cmsis-examples), which has two targets (HE and HP) and two projects (KWS and object detection), so that KWS is available only for HE core and object detection for HP core.

With CMSIS CSolution extension v1.46, user cannot change the active context (project selection does not work), but this works ok with extension version 1.44, so there is some kind of regression between these releases.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone https://github.com/alifsemi/alif_mlek-cmsis-examples and open it with VSCode
2. Go to CMSIS extension
3. Click 'Manage Solution Settings'
4. Try to select 'Alif-E7-M55-HE' and 'kws'

**Expected behavior**
Active context is changed to Alif-E7-M55-HE and kws.

**Code sample and logs**

**Screenshots**
kws is not activated:

[image]

**Additional context**
This is a regression between v1.44 and v1.46.

ReinhardKeil, 2025-01-23:
The `for-context:` in the *.csolution.yml file is causing the issue.  The extension has a problem with it and we will fix this in the release that is planned for end of Jan 2026.

As workaround comment the following lines. 

```yml
projects:
    - project: ./object-detection/object-detection.cproject.yml
#      for-context:
#       - +Alif-E7-M55-HP
    - project: ./kws/kws.cproject.yml
#     for-context:
#       - +Alif-E7-M55-H#175 
```

JonatanAntoni, 2025-01-24:
@KimmoVaisanen-AlifSemi,

Thanks for raising this. The problem is that our current dialog design instantly tries to save each individual change. I.e., once you change the active target selection it tries to change the context set from `object-detection.debug+Alif-E7-M55-HP` to `object-detection.debug+Alif-E7-M55-HE`.

In your scenario the new context `object-detection.debug+Alif-E7-M55-HE` is invalid, because of the `for-context` condition. Hence, the change is discarded. You'd need to make two changes at onces, i.e., change the active target *and* the active project to get another valid context.

We need to check if we can safely reflect the change to the active target in the dialog without actually persisting it. This would result in deactivating the still selected `object-detection` checkbox and activate the not yet selected `kws` checkbox. Next, you can select the `kws` checkbox to complete a new and valid context selection.

closed 2025-01-24 by JonatanAntoni

ReinhardKeil, 2025-01-24:
@JonatanAntoni thanks for your fix.  Please replace the text "No build type for target" with "Excluded for active target"