ARM-software/vscode-cmsis-csolution #177: Record the original source of a layer in the *.csolution.yml

ReinhardKeil, 2025-01-24:
When adding layers it is unclear where the layer came from and when multiple layers are possible the same default directory is used.

This applies to the configuration solution view.
https://github.com/ARM-software/vscode-cmsis-csolution?tab=readme-ov-file#configure-solution-view

jkrech, 2025-04-04:
Alternative approach would be to store this information in the directory that layer is copied to. I was thinking in a similar way to how we hand config files in RTE.

```
$SolutionDir()$/Board/B-U585I-IOT02A/

Keil.B-U585I-IOT02A_BSP.base@2.0.0
```

Within the file the relative path to the layer could be stored e.g. `CMSIS/Board` as layers do not have a name or unique ID.

ReinhardKeil, 2025-05-22:
We will go for the original proposal in https://github.com/ARM-software/vscode-cmsis-csolution/issues/177#issue-2808821371

closed 2025-05-26 by jkrech

ReinhardKeil, 2025-05-26:
This does not work for multiple layers. 

```yml
      variables:
        - Board-Layer: $SolutionDir()$/Board/NUCLEO-H743ZI2/Board.clayer.yml
          copied-from: ${CMSIS_PACK_ROOT}/Keil/NUCLEO-H743ZI2_BSP/1.0.0/Layers/Default/Board.clayer.yml
        - SDSIO-Layer: $SolutionDir()$/sdsio/network/sdsio_network.clayer.yml
          copied-from: $SolutionDir()$/../../../w/SDS-Framework/layer/sdsio/network/sdsio_network.clayer.yml
```

This creates a warning the CMSIS-Toolbox.
```txt
warning csolution: variable 'copied-from' redefined from 'C:/Users/reikei01/AppData/Local/Arm/Packs/Keil/NUCLEO-H743ZI2_BSP/1.0.0/Layers/Default/Board.clayer.yml' to 'C:/Test/May22/SDS4/../../../w/SDS-Framework/layer/sdsio/network/sdsio_network.clayer.yml'
```

Suggest for now to comment this feature and fix first the CMSIS-Toolbox before this is enabled again.