ARM-software/vscode-cmsis-csolution #178: CMSIS view context menu - Solution

mguzmanm, 2025-01-27:
This is a proposal for the context menu in the CMSIS view of the CMSIS Solution extension.

**Solution**
The CMSIS solution level needs an additional item in the context menu:

**Find**
Explanation
VS Code has a find file function built in that we can use for this:

Ctrl - Alt - f (Windows/Linux)
Cmd - Opt - f (MacOS)
This functionality can help to find a file within the CMSIS-View on the solution level.

jkrech, 2025-02-17:
An extension can contribute a default key binding inside `package.json` under `"contributes":` node

To be validated

```
  "contributes": {
    "keybindings": [
      {
        "command": "list.find",
        "key": "ctrl+alt+f",
        "mac": "cmd+alt+f", 
        "when": "view == cmsis-csolution.outline && cmsis-csolution.activeSolutionState==active",
      }
    ]
  }
```

jkrech, 2025-02-21:
I think searching in a View is named "Search" (Menu View -> Search) and is given the default keyboard shortcut: Ctrl-Alt-F
We could add an icon (magnifying glass) if that is not overloading the list or icons and otherwise add "Search" to the `...` menu.

ReinhardKeil, 2025-02-21:
Just to be clear, we should name the feature  "Search" and add it to the drop down menu as shown here.

[image]

Also the command should show the number of found items similar to the Explorer view 

[image]

ReinhardKeil, 2025-02-21:
The key binding is standard, so there should be no action required.

arneschmid, 2025-02-24:
Badges showing the number of matched files underneath a tree node is not possible. The list.find internal command doesn't expose this as a feature and doesn't provide any events to hook in.

Other views like the "Source Control" don't provide this feature, as well.

To implement this, we need to implement a custom Search / Filter widget to override the default one and update the tree and highlighting on our own. The list.find action can't be extended, as well.

jkrech, 2025-02-24:
I propose to implement the feature without searching in collapsed items and not showing the number of matched files in a first step.

arneschmid, 2025-02-27:
We need to add the keybinding to ensure, the hotkey name is also shown in the menu

[image]

Also there is an obsolete command in the package.json having no code behind, throwing an error, when executing