ARM-software/vscode-cmsis-csolution #182: Add "Copy Include File" action button

ReinhardKeil, 2025-01-29:
The "Copy Include File" action button should copy "#include "<filename>/n" to the clip board. It should be on components that provide header files and header files in the CMSIS View as shown in this mockup.

[image]

edriouk, 2025-02-06:
The following should appear in a command handler: 
```
 const incText = `#include "${path.basename(uri.fsPath)}"`;
 await vscode.env.clipboard.writeText(incText)

```

mguzmanm, 2025-02-27:
For release 1.50, it was agreed to do not include copy "#include file" functionality for header files in components section. This functionality will come later after refactoring CMSIS View [#193](https://github.com/ARM-software/vscode-cmsis-csolution/issues/193) Refactor CMSIS View

mguzmanm, 2025-03-31:
Required changes are implemented in PR https://github.com/Arm-Debug/vscode-cmsis-csolution/pull/2200

mguzmanm, 2025-04-01:
Required changes were merged in main branch.

closed 2025-04-01 by mguzmanm