ARM-software/vscode-cmsis-csolution #183: Changes to `cbuild-pack.yml` should trigger `cbuild setup`

ReinhardKeil, 2025-01-30:
To update packs, an effective method is to change the `cbuild-pack.yml` file.   

A change to this file should trigger a `cbuild setup` to reflect new pack versions.

jkrech, 2025-01-31:
Since this file is also updated by the tools if a pack changes in other solution input files we need to make sure that we don't run the `cbuild setup` unnecessarily for a second time.

JonatanAntoni, 2025-02-06:
How can we know? Shall we have a grace period if the file is changed while cbuild is active?

ReinhardKeil, 2025-02-06:
I suggest that during execute of cbuild this file watches are disabled.

jkrech, 2025-02-06:
The safest way would be a dedicated editor view for `*.cbuild-pack.yml` with a safe button that would trigger a `cbuild setup` instead of a file watch going forward.

ReinhardKeil, 2025-02-06:
Agreed, let's park this ticket for now.

jkrech, 2026-03-06:
Evgueni, is this solved by the improved event handling?