ARM-software/vscode-cmsis-csolution #185: clangd requires manual restart after changes to project settings.

ReinhardKeil, 2025-01-31:
Currently it is required to execute "clangd: Restart Language Server" in the following situations:

1. A change to `.vscode/settings.json:
    ```
    "clangd.arguments": [
           "--compile-commands-dir=c:\\Test\\Jan10\\M410_Adc\\ARM\\out\\M410_Adc\\Target_1"
       ]
    }
   ```

JonatanAntoni, 2025-02-06:
I'd not expect these files to be modified manually. I.e., no need to put file watches on them. Instead, I can see that these files are repeatedly updated, e.g., by/after `cbuild setup`.
Doesn't clangd auto-detect changes to its `compile_commands.json`? If yes, could this be seen as a shortcoming/bug in the clangd extension? Hence, any addition at our end would be a workaround.

JonatanAntoni, 2025-02-11:
@brondani: Please check the above statement

> hopefully you're generating the new file (`compile_commands.json`) into a new location and renaming: bad things may happen if we see a half-written file

brondani, 2025-02-11:
> hopefully you're generating the new file (`compile_commands.json`) into a new location and renaming: bad things may happen if we see a half-written file

The final location is the output directory which is always the same, but it is done with the CMake command `${CMAKE_COMMAND} -E copy_if_different` which should be reasonably solid and does not touch the file when there are no changes.

ReinhardKeil, 2025-02-21:
This needs more review from my side.

ReinhardKeil, 2025-05-21:
cannot duplicate anymore

closed 2025-12-16 by jkrech