ARM-software/vscode-cmsis-csolution #187: AC6 symbol export to .clangd file missing

ReinhardKeil, 2025-01-31:
The .clangd file should contain a list of AC6 specific compiler symbols.  

This list is generated using AC6 compiler.

jkrech, 2025-02-18:
@thorstendb-ARM Note, I am seeing the compiler symbols in a project for the STMicroelectronics::STM32U585AIIx (Cortex-M33) but not for an STMicroelectronics::STM32F207IGHx (Cortex-M3). Please check whether there is code executed only for certain processors.

thorstendb-ARM, 2025-02-24:
Found:
- M33 (STM32U5) project runs
- M3 (STM32F200) "Create new Solution" project also runs
- M3 (STM32F200) "UV2CSolution converted" does not run

Difference:
- uv2csolution converted project did not have "build-types" node in csolution.yaml file

jkrech, 2025-02-24:
Good catch. Adding a build-type fixed my project for Cortex-M3 (STMicroelectronics::STM32F207IGHx).
Is there a reason for depending on an existing build-type?

closed 2025-03-03 by thorstendb-ARM