ARM-software/vscode-cmsis-csolution #188: Changes to Features

ReinhardKeil, 2025-02-03:
[image]

ReinhardKeil, 2025-02-03:
Changes to users guide are already reflect in https://github.com/MDK-Packs/vscode-cmsis-solution-docs/commit/db1c752feb5bff86799292aae85f6b715722d4b2

jkrech, 2025-02-03:
@jreineckearm can you please review. 
I am afraid that removing
```
cmsis-csolution.getProcessorName 
```
will create a regression for debugging multi-core targets. 
Today the processor name of the 'active' context is returned.

ReinhardKeil, 2025-02-03:
@jreineckearm what should be returned when more than one context is active?

jreineckearm, 2025-02-03:
@ReinhardKeil , we are aware that this is insufficient for complex use cases. And rather misleading as it is. Yet **_it is used_** with existing debugger configurations for simple contexts. Which will break if we remove this command. We've been there before with other commands that got removed.

ReinhardKeil, 2025-02-04:
@jreineckearm we will not remove getProcessorName at this point it time.  I changed the table.

jreineckearm, 2025-02-04:
Thanks, @ReinhardKeil , let's review this when we get back to proper multi-core debug connections.

JonatanAntoni, 2025-02-05:
| Command  | Change |
|----------|-------------|
| `${command:cmsis-csolution.delete}`                       | Fixed title |
| `${command:cmsis-csolution.getHardwareAndToolchainInfo}`  | removed |
| `${command:cmsis-csolution.getBinaryFile}`                | Changed title |
| `${command:cmsis-csolution.getBinaryFiles}`               | Changed title |
| `${command:cmsis-csolution.getBoardName}`                 | Added command |
| `${command:cmsis-csolution.getBspName}`                   | Changed title |
| `${command:cmsis-csolution.getBspPath}`                   | Changed title |
| `${command:cmsis-csolution.getCbuildRunFile}`             | replacement for getCbuildRunPath |
| `${command:cmsis-csolution.getCbuildRunPath}`             | deprecated |
| `${command:cmsis-csolution.getDeviceName}`                | Changed title |
| `${command:cmsis-csolution.getDfpName}`                   | Changed title |
| `${command:cmsis-csolution.getDfpPath}`                   | Changed title |
| `${command:cmsis-csolution.getProcessorName}`             | deprecated |
| `${command:cmsis-csolution.getSolutionFile}`              | replacement for getSolutionPath |
| `${command:cmsis-csolution.getSolutionPath}`              | deprecated |
| `${command:cmsis-csolution.getTargetPack}`                | deprecated |
| `${command:cmsis-csolution.activateSolution}`             | Changed title |

closed 2025-02-10 by JonatanAntoni

Elise-Kaminski, 2025-02-12:
Discussed with the team today. No changes on the debugger side for now (and no changes for the documentation). The commands are deprecated but still work.

jkrech, 2025-02-17:
build: 1.48.1-14-g1be8064 from main:

- cmsis-csolution.delete - duplicate removed 
- getHardwareAndToolchainInfo - removed
- getSolutionFile - added
- getCbuildRunPath -> getCbuildRunFile
- Select Active Solution from workspace in menu and Title changed.
- Updated Titles validated against table:
finding: 
`${command:cmsis-csolution.getCbuildRunFile}` | Path and name to cbuild-run.yml file of Active Target
of Active Target is missing
fixed in:
https://github.com/Arm-Debug/vscode-cmsis-csolution/pull/2129