ARM-software/vscode-cmsis-csolution #191: CMSIS View: Board and Device Vendor displayed inconsistently

RobertRostohar, 2025-02-04:
Type: Bug Report

**Describe the bug**
- OS and Version: Win11
- VS Code Version: 1.96.4
- Extension Version: 1.48
- Target Device: STMicroelectronics::STM32L4R9AIIx

CMSIS View displays the Board and Device Vendor inconsistently
- Board with vendor
- Device without Vendor

**To Reproduce**
Create a solution with a target which defines a board and device. For example:
```yml
  target-types:
    - type: STM32L4R9I-DISCO
      board: STMicroelectronics::STM32L4R9I-DISCO
      device: STMicroelectronics::STM32L4R9AIIx
```
Open the CMSIS View and check for Board and Device name

**Expected behavior**
Board and Device Vendor should be consistently shown or not show for both rather than a mixture.

**Screenshots**
[image]

jkrech, 2025-02-04:
I assume that the information is fetched from `*.cbuild-idx.yml` files. 
In CMSIS-Toolbox 2.8.0 this issue https://github.com/Open-CMSIS-Pack/devtools/issues/1930 will be fixed and always provide the full deviceID and boardID. The CMSIS View implementation should use the provided deviceID and boardID.

mguzmanm, 2025-04-01:
Required changes are implemented in PR https://github.com/Arm-Debug/vscode-cmsis-csolution/pull/2208

mguzmanm, 2025-04-01:
Required changes were merged in main branch.

closed 2025-04-01 by mguzmanm

jkrech, 2025-04-02:
Validated using internal build: 1.50.1-21-geb8d2c7
Consistently showing vendor for board and device (using CMSIS-Toolbox 2.8.0)
[image]