ARM-software/vscode-cmsis-csolution #193: Refactor CMSIS View

ReinhardKeil, 2025-02-04:
Improve the structure of the CMSIS View

edriouk, 2025-02-04:
The central idea of the refactoring is to replace SolutionNode type in SolutionOutlineTree with CTreeItem class.

Advantages: 
- CTreeItem represents a generic tree with label, attributes and child elements
- new attributes can be introduced without modifying the type
- can be written to a YAML string that can ease testing: compare result with golden reference

An estimated effort is ~2 weeks including adding tests for the existing Outline View content functionality

edriouk, 2025-02-05:
It make sense to use a SolutionOutlineItem extended from CpTreeItem to accommodate additional  functionality, in particular:
- provide specific methods
- store reference to the original CpTreeItem read from YAML file: this can ease editing

mguzmanm, 2025-03-31:
Required changes were merged in main branch.

closed 2025-03-31 by mguzmanm