ARM-software/vscode-cmsis-csolution #200: uv2csolution executes cbuild setup twice

ReinhardKeil, 2025-02-04:
Execute uv2csolution using context menu in Explorer View runs cbuild setup twice.

Use this project: [test.zip](https://github.com/user-attachments/files/18659956/test.zip)

```
uv2csolution.exe c:\Test\Jan10\10872\test.uvprojx
1 out of 1 target(s) converted
Created solution: C:/Test/Jan10/10872/test.csolution.yml
With projects:
- C:/Test/Jan10/10872/test.cproject.yml
Enqueue: cbuild setup c:\Test\Jan10\10872\test.csolution.yml --context-set --packs
Execute: cbuild setup c:\Test\Jan10\10872\test.csolution.yml --context-set --packs
C:/Test/Jan10/10872/test.csolution.yml:2:3 - error csolution: schema check failed, verify syntax

Completed: cbuild failed with exit code 1
Enqueue: cbuild setup c:\Test\Jan10\10872\test.csolution.yml --context-set --packs --update-rte
Execute: cbuild setup c:\Test\Jan10\10872\test.csolution.yml --context-set --packs --update-rte
C:/Test/Jan10/10872/test.csolution.yml:2:3 - error csolution: schema check failed, verify syntax

Completed: cbuild failed with exit code 1
```

JonatanAntoni, 2025-02-06:
Cannot reproduce.
Whats the reason behind the error message
```
C:/Test/Jan10/10872/test.csolution.yml:2:3 - error csolution: schema check failed, verify syntax
```

JonatanAntoni, 2025-02-11:
@mguzmanm, could you have a look into this UV project and if the conversion results are as expected, please?

ReinhardKeil, 2025-02-11:
@JonatanAntoni @mguzmanm we verify this again once the uv2csolution is final.  The problem is that the internal test versions set "version" to "unknown".

JonatanAntoni, 2025-03-13:
In the latest version, there shall be a notification before overwriting an existing solution.
Nevertheless, nothing would stop you from triggering the command multiple times in a row. We'd need to disable the menu or command while uv2csolution is running.

JonatanAntoni, 2025-03-17:
Enhancing the logging of the extension in Arm-Debug/vscode-cmsis-csolution#2171.
Enable trace log via extension setting `cmsis-csolution.logLevel` and locate the log file folder with command `cmsis-csolution.showLogs`.

closed 2025-03-18 by JonatanAntoni

jkrech, 2025-04-03:
validated successfully with 1.50.1-26-g826e668