ARM-software/vscode-cmsis-csolution #204: Clangd choosing the wrong compile commands

MK-HTX, 2025-02-18:
Type: Bug Report

**Describe the bug**
- OS and Version: Windows 10 22h2
- VS Code Version: 1.97.1
- Extension Version: 1.48.0

The following help says 'After you install clangd, you do not need any extra setup.':
https://developer.arm.com/documentation/108029/0000/Get-started-with-an-example-project/Finalize-the-setup-of-your-development-environment/clangd.

But I cannot get it to work correctly for my own CSolution. I have checked and the issue is also present in the SimpleTrustZone project, but due to its simplicity only by inspecting the clangd output window. My own CSolution has different include paths for the for CProject builds, i.e. the non-secure includes RTOS and the secure not. VSCode complains that RTOS related functions are undefined. A check of the clangd output shows that it is choosing the wrong compile commands.

The issue can be reproduced from the CSolution SimpleTrustZone project as follows:

**To Reproduce**
1. Open the CSolution SimpleTrustZone project,
1. Close all files,
1. Open main_s.c,
1. Open the clangd output window,
1. Restart the clangd server.

**Expected behavior**
The clangd extension is using build commands from the 'CM33_s' cproject for file from the 'CM33_s' project.

**Code sample and logs**

The cland output windows show the following:

```
I[17:20:40.959] argv[0]: c:\Users\...\AppData\Roaming\Code\User\globalStorage\llvm-vs-code-extensions.vscode-clangd\install\19.1.2\clangd_19.1.2\bin\clangd.exe
I[17:20:40.959] argv[1]: --compile-commands-dir=c:\repos\csolution-examples\SimpleTrustZone\out\CM33_s\AVH\Debug
I[17:20:40.959] argv[2]: --compile-commands-dir=c:\repos\csolution-examples\SimpleTrustZone\out\CM33_ns\AVH\Debug
```
and:

```
I[17:20:40.999] Loaded compilation database from c:\repos\csolution-examples\SimpleTrustZone\out\CM33_ns\AVH\Debug\compile_commands.json
I[17:20:41.000] ASTWorker building file c:\repos\csolution-examples\SimpleTrustZone\CM33_s\main_s.c version 1 with command inferred from C:\repos\csolution-examples\SimpleTrustZone\CM33_ns\main_ns.c
```
**Additional context**

It can be seen that the compile command folders from both CProject folders are provided to clangd, but it is choosing the wrong compilation options. The AST worker is using a compilation command inferred from a CM33_ns CProject file to build and a CM33_s CProject file.

ReinhardKeil, 2025-03-13:
Potential solution:

Add "Active Clangd information" at the project level.  The activated project should be highlighted (bold or different color).  
[image]

Only one project at a time can be actived.  
The button updates the file `settings.json` with the related database entry.

The icon could be a circle (i) similar to this:
[image]

JonatanAntoni, 2025-03-17:
Relates to https://github.com/clangd/clangd/issues/1092

JonatanAntoni, 2025-03-19:
@MK-HTX, may I ask you to crosscheck my hypothesis, please?

The `--compile-commands-dir` arguments are added via `.vscode/settings.json`. In a multi-project solution there is one argument added per project. But on my Mac using the clangd from Apple Clang installation, only the last one is actually used. And having these arguments given, clangd seems to ignore the compilation database references in the `.clangd` files per project folder.

I deleted the `.vscode/settings.json` (or removed the `clangd.arguments` setting) and restarted clangd. Now, it starts loading the compilation database files listed in the projects `.clangd` files.

The `clangd.arguments` have been added due to clangd/clangd#1875. As this seems to break other scenarios, we might need to find another solution.

MK-HTX, 2025-03-19:
@JonatanAntoni, I can confirm that with the compile commands removed:
- The clangd compilation options for the project related source files look correct, 
- Pack file compilation options are now using clangd defaults and don't even know which target to compile against.

JonatanAntoni, 2025-03-20:
> Pack file compilation options are now using clangd defaults and don't know which target to compile against.

Indeed, that's the misery, here. We need look how this could be solved for both situations.

ReinhardKeil, 2025-04-04:
There should be always one project selected.  Having no project selected means that header files from components coming from packs are not found.

[image]

The command description to use is:
- "Activate Clangd information" (instead of Set Clangd context)
"Clangd information active" (instead of Unset Clangd context)

JonatanAntoni, 2025-04-08:
@Elise-Kaminski, lets discuss required updates to the user guide.

closed 2025-04-08 by jkrech