ARM-software/vscode-cmsis-csolution #205: Add support for installing a CMSIS-Pack in the CMSIS Solution extension using a `URI`

jkrech, 2025-02-21:
Type: Feature Request

Install a CMSIS Software Pack in the CMSIS Solution Extension using a `URI` that can be embedded in a "document" or "webpage"

something like: 
a) packid:  `vscode://arm.cmsis-csolution/installpack?packID=ARM::CMSIS-RTX`
b) download url: `vscode://arm.cmsis-csolution/installpack?url=https://www.keil.com/pack/ARM.CMSIS-RTX.5.8.0.pack`

The CMSIS Solution Extension shall invoke 
a) `cpackget add <packID>`
b) `cpackget add <url>`

Reference: 
Today this URL is used to download and extract an example project from a URL into a location chosen by the user by the Keil Studio Pack extension:
`vscode://arm.keil-studio-pack/zip?url=https://downloads.software.api.keil.arm.com/projects/blinky-bf3ed025.zip`

ReinhardKeil, 2025-02-21:
I would suggest that we use the format (a).  It should allow also to specify (optionally) versions.

jkrech, 2025-02-21:
I am proposing to support both, as users could trigger an installation for non-public packs with it.

jkrech, 2025-03-19:
Using internal build version: 1.50.1-10-g49e010c

Note that github md files do not render uri starting with `vscode` as links
[copy this link into your browser to install `ARM::CMSIS-View` inside vscode](vscode://arm.cmsis-csolution/installpack?pack=ARM::CMSIS-View)

```
vscode://arm.cmsis-csolution/installpack?pack=ARM::CMSIS-View
```

closed 2025-04-01 by jkrech