ARM-software/vscode-cmsis-csolution #206: Missing source files in *.cgen.yml after generating code in CubeMX for a project using Azure RTOS

jenhua01, 2025-02-21:
Type: Bug Report

**Describe the bug**
- OS and Version: Windows 11 Enterprise 23H2
- VS Code Version: 1.97.2
- Extension Version: 1.48.0
- Target Device: STM32F423VHTx (not limited to)

After generating code in CubeMX for an STM32F423VHTx project using Azure RTOS, the middleware source files are missing. Upon inspecting the *.cgen.yml file, we can see that the middleware header files are included:

```
add-path:
    - ...
    - ./STM32CubeMX/Middlewares/ST/threadx/common/inc
    - ./STM32CubeMX/Middlewares/ST/threadx/ports/cortex_m4/ac6/inc
    - ...
```

However, the source files are not, unlike those from CubeMX and STM32 HAL Driver, which are correctly listed in the *.cgen.yml file below.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a CubeMX basic solution for device STM32F423VHTx
2. Open CubeMX and click on 'Middleware and Software Packs' > 'X-CUBE-AZRTOS-F4'
3. Click on 'STMicroelectronics.X-CUBE-AZRTOS-F4' > 'RTOS ThreadX' > 'ThreadX', select 'Core' and press 'Ok'
4. Click on 'Middleware and Software Packs' > 'X-CUBE-AZRTOS-F4' once again and enable the 'RTOS ThreadX'
5. Click on 'GENERATE CODE' and 'Yes'
6. Build the project
7. See the 'Error: L6218E...', since the linker cannot find the corresponding middleware source files

**Expected behavior**
All source files under the .../.../Middleware/ST/threadx/... folder should be listed in the *.cgen.yml file.

**Code sample and logs**
The defective project is attached for reference:
[CubeMX_Basic_solution.zip](https://github.com/user-attachments/files/18906015/CubeMX_Basic_solution.zip)

jenhua01, 2025-02-21:
**Possible workaround**
Need to manually add a source code group in the *.cproject.yml file and include all relevant .c and .s files there, e.g.

```
groups:
    - group: STM32CubeMX Middleware RTX
      files:
        - file: ./STM32CubeMX/STM32F423MHYx/STM32CubeMX/Middlewares/ST/threadx/common/src/txe_block_allocate.c
        - file: ./STM32CubeMX/STM32F423MHYx/STM32CubeMX/Middlewares/ST/threadx/common/src/txe_block_pool_create.c
        - file: ...
    - group: STM32CubeMX Middleware Port
      files:
        - file: ./STM32CubeMX/STM32F423MHYx/STM32CubeMX/Middlewares/ST/threadx/ports/cortex_m4/ac6/src/tx_thread_context_restore.S
        - file: ./STM32CubeMX/STM32F423MHYx/STM32CubeMX/Middlewares/ST/threadx/ports/cortex_m4/ac6/src/tx_thread_context_save.S
        - file: ...
```

This approach is necessary because the *.cgen.yml file will be overwritten if regenerate the code in CubeMX in the future (, if necessary). Now, those missing files should be seen and included in the CMSIS View.

jkrech, 2025-02-21:
I have created a bug against the cbridge tool here:
https://github.com/Open-CMSIS-Pack/generator-bridge/issues/129

ReinhardKeil, 2025-02-21:
As a workaround for the time being, you can add the source files manually. I know that this is not a good solution, but I believe that the ThreadX stack should be in reality an standard CMSIS-Pack that is managed by the VS Code extension rather then CubeMX.

jkrech, 2025-04-24:
Should be fixed in CMSIS-Toolbox 2.9.0

closed 2025-04-24 by jkrech