ARM-software/vscode-cmsis-csolution #209: Create New Solution from template project requires --update-rte

ReinhardKeil, 2025-03-04:
Type: Feature Request

When starting a new solution using a template project, the device, board, and pack is configurable as the template works for different devices.  Therefore `cbuild setup` should be called with `--update-rte`.

To reproduce use this template:
https://github.com/Open-CMSIS-Pack/NXP_FRDM-MCXN947_BSP/blob/main/NXP.FRDM-MCXN947_BSP.pdsc#L1801

JonatanAntoni, 2025-03-10:
We'd need to add another call to `cbuild setup` once the project got created but *before* opening the new solution workspace. Alternatively, we'd need to pass some additional information into the new instance of vscode. @thorstendb-ARM what do you think?

ReinhardKeil, 2025-03-10:
I don't think you need another call. Just in case that the project go initiated via a template, --update-rte should be applied.

JonatanAntoni, 2025-03-10:
As of today, we explicitly do *not* call `cbuild setup --update-rte` after opening a solution workspace. Can we detect from the files present (or not present) in the workspace, that we'd be better off calling `--update-rte`? This might result in the most robust solution. I.e., moving away from the strict rule of not using `--update-rte` at the first `cbuild setup` call.

thorstendb-ARM, 2025-03-11:
>  @thorstendb-ARM what do you think?
As we just discussed, I would sugest to let cbuild run with --update-rte after project creation (prob. just for selected projects)and before project open.
If this can be called with await this should give us a proper sync point to continue with "Open Solution" after cbuild has run, or cancel in case of any error.

JonatanAntoni, 2025-03-11:
At the point in time where we copy the draft project, there is no `cbuild` command available unless the user happens to have a global toolbox installation.
I.e., it looks like we can't run `cbuild setup --update-rte` as part of the project creation. Thus, I am back to use this flag in difference to the normal behaviour on workspace load once we detect an incomplete project. But how to know that the workspace loaded contains an incomplete project? We might leave some magic flag file in the workspace or add an indicator to the `.cbuild-set.yml` file?

thorstendb-ARM, 2025-03-19:
Seems, uv2csolution converted projects also need this step, reproduced running:
`cbuild setup Blinky.csolution.yml --update-rte -S`

Reproduce:
- install Keil::STM32F4xx_DFP@2.17.1
- Create new project: F400 Board, CMSIS-RTOS Blinky
- Open
- Compile

JonatanAntoni, 2025-03-19:
> * install Keil::STM32F4xx_DFP@2.17.1
> * Create new project: F400 Board, CMSIS-RTOS Blinky

This example doesn't ship `RTE/Device/STM32F407IGHx/STM32F405_415_407_417_427_437_429_439.dbgconf` which is expected to be present by cbuild/csolution.

jkrech, 2025-04-01:
Validating build: 1.50.1-19-gbccd9e2
The only trace left in cbuild-set.yml is
```
environment: {}
```
I clearly see that update-rte is invoked. Due to required compiler selection there is another invocation:
```
Execute: cbuild setup c:\workspace\template_project\CM3\Blank.csolution.yml --context-set --packs --update-rte
Completed: cbuild failed with exit code 1
Execute: cbuild setup c:\workspace\template_project\CM3\Blank.csolution.yml --context-set --packs --update-rte
error csolution: processing context 'Project.Debug+ARMCM3' failed
error csolution: compiler undefined, use '--toolchain' option or add 'compiler: <value>' to yml input, selectable values can be found in cbuild-idx.yml

Completed: cbuild failed with exit code 3
Execute: cbuild setup c:\workspace\template_project\CM3\Blank.csolution.yml --context-set --packs --update-rte
+-----------------------------------------------
(1/1) Setting up context: "Project.Debug+ARMCM3"
Completed: cbuild succeed with exit code 0
```

closed 2025-04-01 by jkrech