ARM-software/vscode-cmsis-csolution #210: [csolution] Race condition on cbuild/compile_commands and Intellisense updateClangdConfig() file generation (adds defines)

thorstendb-ARM, 2025-03-07:
**Describe the bug**
Race condition on cbuild/compile_commands and Intellisense updateClangdConfig() file generation (adds defines).

**To Reproduce**
- Open any project (I used github/AWS_MQTT_Demo)
- find and remove any .clangd file(s)
  - (when found the relevant one, open in editor, empty it and save)
- Open CMSIS, click on "Manage Solution Settings"
- Change Active Projects, Build type
  - here: [x] Demo: Debug / Release
- see resulting file: .clangd (Solution or Project dir)
  - -D defines are missing

[image]

**Expected behavior**
.clangd file fully populated with definitions

**Environment (please complete the following information):**
- Arm CMSIS CSolution extension
- Version v1.50.0
- OS: windows, darwin, prob. also linux

**Additional context**
- GUI click ends up in src/views/context-selection/context-selection-webview-main.ts, SET_SELECTED_CONTEXTS -> setSelectedContexts()
- src/solutions/clangd-manager.ts, function updateClangdConfig() gets triggered before compile_commands.json is ready
- Not quite sure where this call is fired, but /Users/thode01/work/github/vscode-cmsis-csolution/src/solutions/cmsis-toolbox.ts, runCmsisTool() Line 177 console.log(msg); hasn't been reached before. Seems there is another sync mechanism other than "solution files changed"

JonatanAntoni, 2025-03-11:
`compile_commands.json` needs to be deleted to reproduce this issue. For updating `.clangd` configuration, the defines and includes are read from the generated `compile_commands.json`. This requires `cbuild setup` to run. Currently, updating `.clangd` is triggered directly by configuration change (i.e., selecting another build context). Instead, we should assume the configuration change will trigger `cbuild setup` and the update of `.clangd` needs to be triggered afterwards.

JonatanAntoni, 2025-03-11:
The cause seems to be concurrently triggering both, cbuild setup *and* clangd updated, via the same events (i.e., context change, and environment activation). Instead, we might want to postpone update of clangd files until cbuild setup completes.

closed 2025-03-18 by jkrech