ARM-software/vscode-cmsis-csolution #212: Software Components change the compiler of .cbuild-run.yml and csolution.yml

sukiElaina, 2025-03-10:
I wan to use AC6@6.22.0，but when i changed the components, compiler settings of the config file  .cbuild-run.yml and csolution.yml changed into gcc

jkrech, 2025-03-10:
Without looking at you CMSIS Solution project and having step by step instructions for reproducing your finding, it is hard to guess what has been going wrong for you.

a) Do you have the AC6 version 6.22.0 installed on  your host machine and the environment variable `AC6_TOOLCHAIN_6_22_0` pointing to the installation's bin folder? If you are using vcpkg in your shell or as part of the VS Code "Arm Tools Environment Manager" extension, you can control this via the `vcpkg-configuration.json` file.

b) In your CMSIS Solution project file `*.csolution.yml` the compiler is configured using the [`compiler:`](https://open-cmsis-pack.github.io/cmsis-toolbox/YML-Input-Format/#compiler) node.
What happens if you manually set `compiler: AC6@6.22.0` in your csolution.yml file? Do you see the correct compiler being used?

BTW: which versions of the CMSIS Solution extension and the CMSIS-Toolbox are you using?

sukiElaina, 2025-03-10:
I use cubemx to generate my project.
this is my `vcpkg-configuration.json`:
```
{
 "registries": [
  {
   "name": "arm",
   "kind": "artifact",
   "location": "https://artifacts.tools.arm.com/vcpkg-registry"
  }
 ],
 "requires": {
  "arm:tools/kitware/cmake": "^3.28.4",
  "arm:tools/ninja-build/ninja": "^1.12.0",
  "arm:compilers/arm/armclang": "6.22.0",
  "arm:tools/open-cmsis-pack/cmsis-toolbox": "^2.7.0",
  "arm:debuggers/arm/armdbg": "6.2.1"
 }
}
```
this is my `csolution.yml` before i change my component:
```
solution:
  created-by: uv2csolution@1.6.0
  created-for: CMSIS-Toolbox@2.7.0
  compiler: AC6@6.22.0
  target-types:
    - type: H743v3
      device: STM32H743IITx
      optimize: debug
  projects:
    - project: H743v3.cproject.yml

```
when i use `Manage software components `,and add dsp library to my project:

[image]

then my `csolution.yml` file was automatically changed to:
```
solution:
  created-by: uv2csolution@1.6.0
  created-for: CMSIS-Toolbox@2.7.0
  compiler: GCC@6.22.0
  target-types:
    - type: H743v3
      device: STM32H743IITx
      optimize: debug
  projects:
    - project: H743v3.cproject.yml
```
this need to manually set `compiler: AC6@6.22.0` and then my project can work well.

sukiElaina, 2025-03-10:
my cmsis solution extension version is 1.50.0
my vscode version is 1.97.2 for windows

jkrech, 2025-03-10:
You generated a uvprojx project and converted it to the CMSIS Solution project format. Did you notice any error message during the conversion?

```
0 out of 1 target(s) converted
Error: 1 error(s), 1 warning(s):
Errors:
	The target "STM32H743IITx" is configured to use Arm Compiler 5.
	CMSIS csolution projects support Arm Compiler 6 only.
	For details on how to manually migrate to AC6, see
	https://developer.arm.com/documentation/100068/0621/Migrating-from-Arm-Compiler-5-to-Arm-Compiler-for-Embedded-6
Warnings:
	The target "STM32H743IITx" is configured without pack's identifier
```

I have opened the project in uVision and configured "Default Arm Compiler 6", stored the project and invoked the conversion again.

Afterwards I opened the Component View and selected the `CMSIS:DSP` component but the compiler remained configured to AC6.

sukiElaina, 2025-03-10:
my uvprojx project can be converted without any error.  But i configure my project  `Arm Compiler v6.22`

sukiElaina, 2025-03-10:
> You generated a uvprojx project and converted it to the CMSIS Solution project format. Did you notice any error message during the conversion?
> 
> ```
> 0 out of 1 target(s) converted
> Error: 1 error(s), 1 warning(s):
> Errors:
> 	The target "STM32H743IITx" is configured to use Arm Compiler 5.
> 	CMSIS csolution projects support Arm Compiler 6 only.
> 	For details on how to manually migrate to AC6, see
> 	https://developer.arm.com/documentation/100068/0621/Migrating-from-Arm-Compiler-5-to-Arm-Compiler-for-Embedded-6
> Warnings:
> 	The target "STM32H743IITx" is configured without pack's identifier
> ```
> 
> I have opened the project in uVision and configured "Default Arm Compiler 6", stored the project and invoked the conversion again.
> 
> Afterwards I opened the Component View and selected the component but the compiler remained configured to AC6.`CMSIS:DSP`

sukiElaina, 2025-03-10:
i use default compiler version 6 and  repeat what I did before, the compiler of csolution.yml can be remained AC6.

closed 2025-03-10 by sukiElaina