ARM-software/vscode-cmsis-csolution #213: Rework View: Software Components

ReinhardKeil, 2025-03-14:
The view "Software Components" allows to select components and packs.

Components sub-view 

[image]

Software Packs

[image]

ReinhardKeil, 2025-06-20:
This is feedback to internal version 1.54.1-123-geff7b80

Test Project: [Hello.zip](https://github.com/user-attachments/files/20796123/Hello.zip)
Test Pack: [ACME.Test.1.1.1.zip](https://github.com/user-attachments/files/20796163/ACME.Test.1.1.1.zip)

ReinhardKeil, 2025-06-27:
Component Variants cannot be selected

jkrech, 2025-07-01:
Status Update: 2025-07-01 - CMSIS Solution VSIX Window: 1.54.1-201-ga121d7a
ships cmsis-toolbox nightly - csolution 2.10.0+p10-g36285395

### I1B: CMSIS-Toolbox (nighly) included in vsix
### I2B: Solutions with layers open Component View and correctly displays all selected components
### I3B: system keeps remembering previous projects, some projects do not open - cannot be reproduced anymore

### I4M: component no longer selected after reopen (clean state on open)
### I5M: after adding or removing a pack the Component View is not refreshed - Components show correctly on reopening the View (ToDo)
### I6L: New Solution Dialog depends cmsis.exe for retrieving information from installed packs: boards, devices, examples, templates, reference apps - ToDo (later)

### C1B: Installing and deleting packs is only reflected in Component View after reopening solution
### C2B: ToDo

###E1M: ToDo
###E2M: In the Component View only components from the currently selected bundle are displayed. The information from the tool tip comes from csolution and is ToD

### Y1B: no more duplicates created but newly added components have <vendor> (no version) - fixed
### Y2M: fixed
### Y3L: ToDo

closed 2025-07-04 by jkrech

jkrech, 2025-07-04:
Status Update 2025-07-04:

C2B: done

[image]