ARM-software/vscode-cmsis-csolution #215: Integration with STM32CubeMX

DBizz22, 2025-03-21:
Error
```
Execute: cbuild ....\sample\CubeMX_Basic_solution\CubeMX.csolution.yml --context-set --packs
.../STM32CubeMX/STM32G431CBUx/CubeMX.cgen.yml - error csolution: cgen file was not found, run generator 'CubeMX' for context 'CubeMX.Debug+STM32G431CBUx'
```

STM32CubeMX is not generating the .cgen.yml file required

DavidLesnjak, 2025-03-21:
Hi @DBizz22,
Could you provide more information:

- Is `STM32CubeMX` successfully launched from `VSCode`?
- Is MX_Device.h generated?
- Did you do any changes in `STM32CubeMX` under `Project Manager` tab? which?

Thanks

DBizz22, 2025-03-21:
- STM32CubeMX was successfully launched from vscode
- MX_Device.h was not generated
- I made a change to generate the code using only the LL library

DBizz22, 2025-03-21:
I discovered when the RCC driver is generated using the LL library, the required files like cgen.yml and MX_Device.h are not generated. Switching the driver to the HAL library fixed it.

closed 2025-03-21 by DBizz22

DavidLesnjak, 2025-03-21:
Thank you for the additional information. I can confirm that the issue is reproducible when using RCC with LL.
A fix is being tracked in [#142](https://github.com/Open-CMSIS-Pack/generator-bridge/issues/142)