ARM-software/vscode-cmsis-csolution #217: "Select Compiler" configuration causes modification of defines in csolution.yml

VladimirUmek, 2025-03-27:
Type: Bug Report

**Describe the bug**
- OS and Version: Windows 11
- VS Code Version: 1.98.2
- Extension Version: Arm CMSIS Solution 1.50.0

I have a csolution.yml with select compiler options (and without compiler selected):
```yml
solution
  select-compiler:
    - compiler: AC6
    - compiler: GCC
    - compiler: CLANG
    - compiler: IAR
```
This same solution file has target types:
```yml
solution:
  target-types:
    - type: CM33
      device: ARMCM33
      define:
        - NS_APP_START_ADDRESS: 0x00200000
      variables:
        - Target-Layer: ../../../Target/CM33/Target.clayer.yml
```
When I open this solution with VS Code, CMSIS Solution extensions opens "Configure Solution" tab with "Select Compiler" options to select compiler. After I click "OK" my csolution.yml gets modified:
- compiler node is added
- all defines in target type get modified

The target type gets modified to:
```yml
    - type: CM33
      device: ARMCM33
      define:
        - NS_APP_START_ADDRESS: "2097152"
      variables:
        - Target-Layer: ../../../Target/CM33/Target.clayer.yml
```
**To Reproduce**
Use [CMSIS-FreeRTOS Examples](https://github.com/ARM-software/CMSIS-FreeRTOS/tree/bb1d1243182161ebad0ce0171c59c4351ec98fe9) to reproduce the behavior. Open "Examples" directory with VS Code.

**Expected behavior**
Only compiler node shall be added, the rest of the solution file shall be unmodified.

JonatanAntoni, 2025-04-03:
@VladimirUmek, as already suspected, this conversion is done by the yaml parser. I.e., it detects `0x00200000` as a valid integer value and discards the formatting information. On write the integer is then serialized in decimal format.

As a workaround, one could wrap the value in double quotes like `"0x00200000"`. This prevents the yaml parser to convert the value to integer and keep it as string, instead.

For a permanent fix, we'd need to decide how we want to handle numbers.

edriouk, 2025-04-04:
However I do not know which parser is used to read/write the file: ours or the Cambridge's one.

edriouk, 2025-04-06:
Fixed in PR https://github.com/Arm-Debug/vscode-cmsis-csolution/pull/2215 
It ensures all numbers are stored as strings in the original format.

This however does not fix the issue entirely, since YAML library forces quotes to the output strings representing numbers.
Changes suggested by ChatGPT do not solve the problem since the emitter adds quotes afterwards and there is no way to alter that behavior.
Thus the fix ensures that only quotes added to the original format by save: 
```
define:
        - NS_APP_START_ADDRESS: 0x00200000
 ```
becomes
```
define:
        - NS_APP_START_ADDRESS: "0x00200000"
```
further saves do not change the quotes

closed 2025-04-08 by jkrech

VladimirUmek, 2025-04-08:
Is current fix treated as temporary workaround?

Note that the value of the define is also passed to the compiler executable as argument in the command string. Compiler will therefore also be invoked with something like
```
armclang --target=arm-arm-none-eabi -DARMCM33 ... -DNS_APP_START_ADDRESS=2097152 ...
```
which may be undesired as it is harder to read.

VladimirUmek, 2025-04-08:
@brondani I understand now that what I am mentioning in my last comment is related to cbuild. Do I raise new issue in CMSIS-Toolbox repo?

brondani, 2025-04-08:
> I understand now that what I am mentioning in my last comment is related to cbuild. Do I raise new issue in CMSIS-Toolbox repo?

@VladimirUmek Yes the hexadecimal value is converted to decimal by `cbuild2cmake`. Please raise a new issue in CMSIS-Toolbox or in cbuild2cmake repo.

edriouk, 2025-04-08:
I have merged new fix for VS code extension:
https://github.com/Arm-Debug/vscode-cmsis-csolution/pull/2226
This ensures that format for numbers (decimal and hex as well as float and boolean values is preserved.
No quotes are added.
Note: This is not related to the cbuild issue