ARM-software/vscode-cmsis-csolution #219: Improve `.gitignore` file

ReinhardKeil, 2025-04-02:
When a project does not provide a `.gitignore` there should be an easy way to add it (for example via a context menu).

The content of the `.gitignore` file needs review as it does not reflect the latest files that should be excluded, but contains files that should be not excluded.

jkrech, 2025-04-02:
If you open the "Source Control" View, select a file and open the context menu you can select "Add to .gitignore". It opens the existing or new file in the editor.
If the current workspace folder is not part of a git repository you can initialize it.

KeilChris, 2025-04-02:
Where do we have a comprehensive list of files not to include in .gitignore?

ReinhardKeil, 2025-04-02:
> Where do we have a comprehensive list of files not to include in .gitignore?

This needs to be done (and reviewed).  This is the current content [.gitignore.txt](https://github.com/user-attachments/files/19569856/default.gitignore.txt)

> "Add to .gitignore". 
This is not the feature I'm looking for. I'm asking for a way to add a default `.gitignore` file if missing.

jkrech, 2025-04-03:
There is a default gitignore file already kept with the CMSIS Solution Extension here:
https://github.com/Arm-Debug/vscode-cmsis-csolution/blob/main/templates/git/.gitignore
We need to
a) review and update it to the latest CMSIS Toolbox build files and default folders
b) need to understand if and when it is used
c) add a context menu to add it

KeilChris, 2025-07-02:
LGTM!

closed 2025-07-04 by jkrech