ARM-software/vscode-cmsis-csolution #220: Add command `cmsis-solution.getSvdFile`

ReinhardKeil, 2025-04-03:
Some extensions or tools may need access to the SVD file of the current target device.

Therefore we should add a new command 

`cmsis-solution.getSvdFile`   Path and name of the SVD file of Active Target

ReinhardKeil, 2025-04-03:
This seems not to work for every device, as devices may have SVD files that are specific to a pname (processor).

Therefore, this ticket is parked.  The proper way is to use the cbuild-run.yml file which can be already accessed.

jkrech, 2025-04-15:
Note that this has been addressed in the CMSIS Debugger extension which now reads this information from the cbuild-run.yml file.
https://github.com/Open-CMSIS-Pack/vscode-cmsis-debugger/pull/151

closed 2025-04-15 by jkrech