ARM-software/vscode-cmsis-csolution #229: CMSIS View Action buttons

ReinhardKeil, 2025-04-23:
The CMSIS View should offer the following action buttons for CMSIS managed targets:

- **Load & Run** a *csolution* application which downloads and starts the application images in the target.
- **Load & Debug** a *csolution* application which downloads the application images and starts the debugger.

The action button:
- **Load & Run** executes from tasks.json the command `CMSIS Load+Run`.
- **Load & Debug** executes from launch.json the first `"request": "attach"` with `cmsis:` section. If not present from tasks.json the command `CMSIS Load+Debug`.

Further commands are available under `...`:
- **Load** executes from tasks.json the command `CMSIS Load`.
- **Erase** executes from tasks.json the command `CMSIS Erase`.
- **Run** executes from tasks.json the command `CMSIS Run`.

[image]

ReinhardKeil, 2025-05-12:
Action buttons are better specified here https://github.com/ReinhardKeil/vscode-cmsis-solution-docs/blob/main/docs/conf_debug.md#cmsis-view---action-buttons

JonatanAntoni, 2025-05-15:
Arm-Debug/vscode-cmsis-csolution#2250