ARM-software/vscode-cmsis-csolution #230: Use "Cmsis-csolution: Experimental Features" to enable target-set

ReinhardKeil, 2025-04-23:
As far as I know, there are currently no "Experimental Features".

Therefore, I suggest to use this setting
[image]

To enable:
  - `-a` option for cbuild/csolution instead of `-S`  
  - new update process for launch.json/tasks.json (https://github.com/ARM-software/vscode-cmsis-csolution/issues/228)

Change the descriptive text from "Enable experimental features" to "Enable target-set instead of context-set (requires CMSIS-Toolbox version 2.9)".

ReinhardKeil, 2025-05-12:
The latest idea is to use `target-set:` in csolution.yml.

See note in https://github.com/ReinhardKeil/vscode-cmsis-solution-docs/blob/main/docs/conf_debug.md

JonatanAntoni, 2025-05-15:
Arm-Debug/vscode-cmsis-csolution#2247

closed 2025-05-19 by jkrech