ARM-software/vscode-cmsis-csolution #231: Remove from Settings: "Cmsis-csolution: Actions"

ReinhardKeil, 2025-04-23:
To my knowledge there is no real need for the setting "Cmsis-csolution: Actions". Therefore this should be removed.

[image]

jkrech, 2025-04-24:
@jreineckearm it is unclear to me when a user would specify their own run and debug configuration in the settings.
see https://github.com/Arm-Debug/vscode-cmsis-csolution/blame/c9fbdf5348601adc8877fb19eb4eb91f4e873aa6/package.json#L1049

jreineckearm, 2025-04-24:
@jkrech , I can only speculate that this was supposed to increase visibility of where run/debug action associations are stored.
The only operation I currently see not covered by the Solution Management dialog is to unlink a button, i.e. clear a config selection.

JonatanAntoni, 2025-05-22:
I'll set this setting as deprecated for the time being. This results in
- The setting is not shown in the UI anymore
- The setting is highlighted with a deprecation message in the json editor
- Until we rework the manage solution dialog, there is still (dead) code referring to this setting