ARM-software/vscode-cmsis-csolution #235: Rework Manage Solution dialog

ReinhardKeil, 2025-05-12:
Type: Feature Request

The Manage Solution dialog is changed as shown below.  

- The user selection of Projects, Images, and Debugger for Context Set is stored in `csolution.yml`.
- The active solution and active `target-set` in VS Code is stored in Code\User\workspaceStorage.

[image]

### Changes

- **Apply** makes the dialog stateful (similar to new Components/Packs dialog). 

### Projects, Images, and Debugger for Target-Set

Allows to choose the projects and images that belong to a `target-set`. 

- **Processor:** is only shown for projects that contain a `pname` selection.
    - `cbuild-idx.yml` and `cbuild.yml` should be extended with `pname:` information.
  
- **Load** allows to select how the project is used. The selection is stored as part of `target-set` in `*.csolution.yml`
    - `Image & Symbols` (default) debug adapter loads debug (DWARF) information and project output image.
    - `Symbols` debug adapter loads only debug (DWARF) information.
    - `Image` debug adapter loads only project output image.
    - `None` debug adapter does not use the output, however the project is included in build.
    - Offset can be specified in the `cproject.yml` file or for images in the `csolution.yml` file. It is shown when specified (but cannot be modified)
    - **Additional Images** shows additional image files that are specified in `csolution.yml` using [`images:`](YML-Input-Format.md#images).

ReinhardKeil, 2025-07-14:
This is a version that adds debug services for discussion.

[image]

Related to https://github.com/Open-CMSIS-Pack/vscode-cmsis-debugger/issues/397

KeilChris, 2025-07-14:
I think we should use the same location for the save button as we have in the manage software components dialog (beginning of the line).

arneschmid, 2025-07-25:
### Proposal: Store workspace-related local user settings in `.vscode/cmsis.json`

This change introduces a dedicated `cmsis.json` file inside the `.vscode` folder of the opened workspace. The file is intended for storing workspace-specific, local user settings.

**Justification:**

- Keeps user-specific configuration separate from project source files, reducing the risk of accidental commits of personal or machine-specific settings.
- Follows the established convention in VS Code extensions to use the `.vscode` folder for workspace-local configuration, making it familiar and discoverable for users and contributors.
- Ensures that settings such as selected targets, UI preferences, or other local choices are not shared or exposed to other users or environments, supporting privacy and individual workflows.
- Prevents workspace-specific settings from polluting the repository and causing merge conflicts or confusion for other contributors.
- The `.vscode/cmsis.json` file can be easily added to `.gitignore`, ensuring it is never checked into version control.
- Provides a flexible foundation for future workspace-local features, allowing us to expand the settings stored here without impacting project files or global extension state.

jkrech, 2025-07-28:
To be more specific, at this point store the least recently selected `target-set` for each `solution` in the workspace.
If no entry exists for a `solution` the default `target-set` is used.

arneschmid, 2025-09-02:
re-opened as working on this isn't finished yet. We need to deprecate cbuild-set and replace with csolution rpc as well as add the debugger services