ARM-software/vscode-cmsis-csolution #240: Add Debuggers

ReinhardKeil, 2025-05-19:
Type: Feature Request

Register CMSIS-DAP Debuggers in https://github.com/Open-CMSIS-Pack/debug-adapter-registry for:

 - Infineon
 - Microchip
 - Nuvoton
 - NXP

The debuggers can be registered using PDSC files.  We need to ensure that these debuggers can be selected seamless from the GUI.

See also: https://github.com/Open-CMSIS-Pack/devtools/issues/2085

jkrech, 2025-05-20:
Searching `$CMSIS_PACK_ROOT`/.Web for `<vendor>` pdsc files with `<debugInterface>` element:
Infineon:
```
adapter="J-Link Lite"
adapter="J-Link"
adapter="XMC-Link"
adapter="JTAG/SW" 
```
Microchip: none
Nuvoton: 
```
Nuvoton.NuMicro_M460_BSP.pdsc:      <debugInterface adapter="Nu-Link"
```
NXP:
```
NXP.QN908x_DFP.pdsc:      <debugInterface adapter="JTAG/SW" connector="10 pin Cortex debug (0.05 inch connector)"/>
NXP.QN908x_DFP.pdsc:      <debugInterface adapter="OpenSDA" connector="Micro-USB"/>
```

KeilChris, 2025-05-21:
Closing with https://github.com/Open-CMSIS-Pack/debug-adapter-registry/pull/7

closed 2025-05-21 by KeilChris