ARM-software/vscode-cmsis-csolution #241: Create target-set: instead of cbuild-set.yml

ReinhardKeil, 2025-05-20:
Type: Feature Request

When `cbuild-idx.yml` contains csolution version 2.9.0 or higher, the Manage Solution dialog should create `target-set:` in  `csolution.yml` instead of the cbuild-set.yml.

An example for this section is in https://open-cmsis-pack.github.io/cmsis-toolbox/YML-Input-Format/#target-set

When `csolution.yml` contains a `target-set:` the `cbuild-set.yml` file should be no longer used by the extension.

JonatanAntoni, 2025-05-22:
Addressed by Arm-Debug/vscode-cmsis-csolution#2266

jkrech, 2025-05-26:
The following scheme got implemented:
- Hide "Run and Debug" settings in "Manage Solution" dialog
- Update target-set in .csolution.yml file to match context selection from "Manage Solution" dialog if `target-set:` exists. (`images:`) node. Please note that no `debugger:` section is written, even if an onboard debugger is recognized and listed in `cbuild-run.yml`.
- Note: `*.cbuild-set.yml` is kept in sync at this point for compatibility/consistency reasons.

closed 2025-05-26 by jkrech