ARM-software/vscode-cmsis-csolution #243: SDS RefApps have a fixed target-type for simulation

ReinhardKeil, 2025-05-21:
The SDS RefApps have a fixed target-type in addition to the flexible target-type for RefApps.

This does currently not work in the VSCode extension as all target-types and not just the first target-type is touched.

The RefApp csolution.yml is defined as follows:
https://github.com/ARM-software/SDS-Framework/blob/ref_apps/examples/usb/SDS_USB.csolution.yml#L18

The expected result (when using the B-U585I-IOT02A board) is:
```yml
  target-types:
    - type: STM32U585AIIx
      variables:
        - SDSIO-Layer: $SolutionDir()$/sdsio/usb/sdsio_usb.clayer.yml
        - Board-Layer: $SolutionDir()$/Board/B-U585I-IOT02A/Board.clayer.yml
      device: STM32U585AIIx
      board: B-U585I-IOT02A:Rev.C

    - type: AVH-SSE-300 # simulator target
      board: ARM::V2M-MPS3-SSE-300-FVP
      device: ARM::SSE-300-MPS3
      define:
        - SIMULATOR
      variables:
        - Board-Layer: $SolutionDir()$/Board/Corstone-300/Board.clayer.yml
        - SDSIO-Layer: $SolutionDir()$/sdsio/fvp/sdsio_fvp.clayer.yml
```

To duplicate use:
https://github.com/ARM-software/SDS-Framework/tree/ref_apps

Then register the local pack with:
>cpackget add <path>/ARM.SDS.pdsc

Use VS Code and start a project as shown here:

ReinhardKeil, 2025-05-22:
After fixing also #177 this is how the target-type list should look like:

```yml
  target-types:
    - type: STM32U585AIIx
      variables:
        - SDSIO-Layer: $SolutionDir()$/sdsio/usb/sdsio_usb.clayer.yml
        - Board-Layer: $SolutionDir()$/Board/B-U585I-IOT02A/Board.clayer.yml
          copied-from: ${CMSIS_PACK_ROOT}/Keil/B-U585I-IOT02A_BSP/2.0.0/CMSIS/Board/
      device: STM32U585AIIx
      board: B-U585I-IOT02A:Rev.C

    - type: AVH-SSE-300 # simulator target
      board: ARM::V2M-MPS3-SSE-300-FVP
      device: ARM::SSE-300-MPS3
      define:
        - SIMULATOR
      variables:
        - Board-Layer: $SolutionDir()$/Board/Corstone-300/Board.clayer.yml
        - SDSIO-Layer: $SolutionDir()$/sdsio/fvp/sdsio_fvp.clayer.yml
```

ReinhardKeil, 2025-05-26:
It is working, but the formatting is a bit weird.  Can the empty line be removed? 

```yml
  target-types:
    - type:

        STM32H743ZITx
      device: STM32H743ZITx
      board: NUCLEO-H743ZI2:Rev.E
      variables:
        - Board-Layer: $SolutionDir()$/Board/NUCLEO-H743ZI2/Board.clayer.yml
          copied-from: ${CMSIS_PACK_ROOT}/Keil/NUCLEO-H743ZI2_BSP/1.0.0/Layers/Default/Board.clayer.yml
        - SDSIO-Layer: $SolutionDir()$/sdsio/network/sdsio_network.clayer.yml
          copied-from: $SolutionDir()$/../../../w/SDS-Framework/layer/sdsio/network/sdsio_network.clayer.yml
```

This is what I would expect

jkrech, 2025-05-26:
Let's not discuss. Regardless whether the `type:` node is followed by an empty line or not, the expectation is that the type name is immediately following the node.

thorstendb-ARM, 2025-05-26:
After investigation with @edriouk we found out, that the YAML reader is faulty wrt newlines.
To easily fix this, please add ${Name} to 'type' node, which will also correctly preserve comments on the same line:
[image]

Result:
[image]

jkrech, 2025-05-26:
```
  target-types:
    - type: ${Name}
```
results in proper formatting

closed 2025-05-26 by ReinhardKeil