ARM-software/vscode-cmsis-csolution #246: Review/Improve Create New Solution flow

ReinhardKeil, 2025-05-27:
Type: Feature Request

The Create New Solution flow needs review:

- tasks.json/launch.json should not contain `arm-debugger` entries by default
- default debugger from PDSC should be selected and entries for CMSIS Debugger generated (when this extension is installed)
- After creating the project CMSIS View should be active (currently Explorer View is active)

JonatanAntoni, 2025-06-17:
> tasks.json/launch.json should not contain arm-debugger entries by default

These seem not generated by our extension.

> default debugger from PDSC should be selected

This is the implicit default. I.e., if an example is shipped without a debugger selection, CMSIS-Toolbox selects the default based on the PDSC file. Once we get the new dialog, we can pre-select the default debugger and update the csolution file on save.

> After creating the project CMSIS View should be active (currently Explorer View is active)

That can be reproduced. Needs to be investigated how to activate a specific view based on the workspace folder content.

ReinhardKeil, 2025-07-07:
Last issue "After creating the project CMSIS View should be active (currently Explorer View is active)" will be addressed with #285