ARM-software/vscode-cmsis-csolution #253: cmsis-toolbox 2.9.0 creation of STM32CubeMX fail, absolute path problem

FerdinandEn, 2025-05-30:
Type: Bug Report

**Describe the bug**
- OS and Version: Win 11 24H2
- VS Code Version: 1.100.2
- "arm:tools/open-cmsis-pack/cmsis-toolbox": "2.9.0"
- Extension: Arm CMSIS Solution 1.54.0/1.52.0

**To Reproduce**
Create Solution for Device STM32F756ZGTx
Solution Template: STM32CubeMX

**_First Start_** of CubeMX from CMSIS View to **_generate a new ioc-File_** fails
- STM32CubeMX starts normal
- at the end of start process an error message occurs in a popup window
    error when loading STMicroelectronics::STM32F756ZGTx.xml file (see screen shot)

[image]

Workaround: Change "arm:tools/open-cmsis-pack/cmsis-toolbox":  to 2.8.0 in vcpkg-configuration.json
            creation of a new project is possible
			
If the STM32CubeMX-File (ioc) exists, it is possible to open STM32CubeMX using cmsis-toolbox version 2.9.0 and modify settings.

**Second problem:**

If "arm:tools/open-cmsis-pack/cmsis-toolbox": "2.9.0" is selected in vcpkg-configuration.json. 
Generation of Code by STM32CubeMX leads to absolute path names in file CubeMX.cgen.yml. Now I got warning messages due to absolute path

If I switch back to cmsis-toolbox version 2.8.0 the path information is relative.

**Project sample**

[TestF756.zip](https://github.com/user-attachments/files/20518455/TestF756.zip)

jkrech, 2025-06-02:
@bgn42 Both problems probably are caused by `cbridge` shipped as part of CMSIS-Toolbox. Can you please investigate and create issues accordingly.

ReinhardKeil, 2025-07-07:
This problem is solved with CMSIS-Toolbox 2.10 - hence closing it.

closed 2025-07-07 by ReinhardKeil