ARM-software/vscode-cmsis-csolution #254: How can I continue to use it after updating to v1.54.0? Flash Device and Arm Debugger

flair0626, 2025-05-31:
I can't find the Run and Debug option in the Manage Solution interface. How can I continue to use the Flash Device?
Use Arm Debugger instead of pyocd. I encountered the following error when using pyocd:

正在执行任务: pyocd erase --probe cmsisdap: --chip --cbuild-run d:\MCU\temp\Blank_solution\Blank_solution+GD32F330F8.cbuild-run.yml
usage: pyocd [-h] [-V] [--help-options] ...
pyocd: error: unrecognized arguments: --cbuild-run

终端进程“C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command pyocd erase --probe cmsisdap: --chip --cbuild-run d:\MCU\temp\Blank_solution\Blank_solution+GD32F330F8.cbuild-run.yml”已终止，退出代码: 1。
终端将被任务重用，按任意键关闭。

ReinhardKeil, 2025-06-02:
Run and Debug is for the new [CMSIS Debugger extension](https://marketplace.visualstudio.com/items?itemName=Arm.vscode-cmsis-debugger).  You need to install this extension which also contains pyOCD.

Previous launch.json scripts should still work.

FerdinandEn, 2025-06-06:
Can I use the st-link with the new debugger?

FerdinandEn, 2025-06-06:
What changes do I have to make in tasks.json and launch.json so that I can start the debugger? At the moment I get the following error message

**Waiting for a debug probe matching unique ID 'ST-LINK' to be connected...**

respectively

**Waiting for a debug probe matching unique ID 'cmsisdap:' to be connected...**

ReinhardKeil, 2025-06-06:
What Host OS are you using?

Please use the command "pyOCD list" in the terminal window to list connected debug adapters.  What is the output with this command.

FerdinandEn, 2025-06-06:
I use Windoes 11.

  #   Probe/Board     Unique ID                  Target
------------------------------------------------------------------
  0   STM32 STLink    0668FF485251667187092118   ✖︎ stm32f756zgtx  
      NUCLEO-F756ZG

ReinhardKeil, 2025-06-06:
- Could you post the launch.json and tasks.json file from the .vscode folder?
- Is there any serial number entered?

FerdinandEn, 2025-06-06:
[launch.json](https://github.com/user-attachments/files/20624123/launch.json)

[tasks.json](https://github.com/user-attachments/files/20624124/tasks.json)

KeilChris, 2025-06-06:
In your `csolution.yml`, insert the following block in `- target-types\- type` section:

```yml
      target-set:
        - set: 
          debugger:
            name: ST-LINK
```

That will preserve the settings if you rebuild the solution or change anything.

FerdinandEn, 2025-06-06:
With the changes in the csolution file, everything works as expected. I can start the debugger.  However, I still get the error messages shown in the log in the debug console. I still cannot see the content of global variables. I am willing to make my project available. 

Should I create a new issue? If so, where?

I have since removed the arm debugger entries.

ReinhardKeil, 2025-06-06:
Are you using AC6? Do you create a HEX file also, using the outputs: node in the cproject file.  
This is to bypass a [GNU Loader Issue](https://open-cmsis-pack.github.io/cmsis-toolbox/Troubleshooting/#gnu-loader-issues-with-arm-compiler-6)

See https://github.com/Arm-Examples/Hello_B-U585I-IOT02A/blob/main/hello.cproject.yml#L38

FerdinandEn, 2025-06-06:
* I use AC.

* I have saved the error messages in a previous post DebugLog.txt.

[DebugLog.txt](https://github.com/user-attachments/files/20624635/DebugLog.txt)

* I have customized cproject

[CubeMX.cproject.yml.txt](https://github.com/user-attachments/files/20625385/CubeMX.cproject.yml.txt)

* If I replace 

 "program": "out/CubeMX/STM32F746ZGTx/Debug/CubeMX.hex",
 
in launch.json, the program is loaded correctly, but the debug symbols are missing, e.g. main.

[launch.json](https://github.com/user-attachments/files/20625477/launch.json)

* In my program there is also the settings.json file.  Do I have to or should I delete the entry for "cmsis-csolution.actions": In my opinion, this refers to the Aem debugger.

[settings.json](https://github.com/user-attachments/files/20625511/settings.json)

FerdinandEn, 2025-06-06:
warning: (Internal error: pc 0x80009d4 in read in CU, but not in symtab.)

The warning appears to be generated when an attempt is made to read the content of a global variable by the debugger. For example, if I enter a variable in the watch window or if a global variable is visible at a breakpoint in the function.

ReinhardKeil, 2025-06-06:
The debug output says, that you are using an Cortex-M7 r0p1 core.  This core has an issue with interrupts that are not disabled.
See https://github.com/Open-CMSIS-Pack/vscode-cmsis-debugger/issues/242

@KeilChris did you document the work-around already?

KeilChris, 2025-06-06:
@FerdinandEn, can you check this file, please: https://github.com/MDK-Packs/RDM2025/blob/main/Blinky/.cmsis/Blinky%2BNUCLEO-F756ZG.dbgconf? This contains the fix for the Cortex-M7 r0p1 core that is on that device.

FerdinandEn, 2025-06-06:
The warnings have disappeared since I added the following entry:
```
  misc:
    - for-compiler: AC6
      C-CPP:
        - -gdwarf-5
      ASM:
        - -gdwarf-5
      Link:
        - --entry=Reset_Handler
```
I am now using the axf file again.

However, I still have the problem that I cannot see the current content of the global variables (axf and hex). In the Watch window as well as when hovering over a global variable, I only ever get the initial value.

@KeilChris I currently only use the systick timer for periodic interrupts. It seems to be stopped.

FerdinandEn, 2025-06-06:
I have found the problem why the content of global variables is not displayed correctly. If the data cache is activated, the content of global variables is not displayed correctly. There are very few variables in my mini-example. That's probably why global variables are all in the cache.

I have checked this for the
* STM32H756ZGTx
* STM32H743ZITx

The problem occurs when the DCache is switched on. The display works without Dcache.

I have never noticed this with the uVision, CubeIDE or STM32Cube for VSC debugger.

dwarf-5 must also be switched on for the STM32H743 to avoid the debugger warning messages.

FerdinandEn, 2025-06-08:
I have found the solution for how to set the debugger so that variables are displayed correctly when the data cache is activated.

Der Wert hprot muss auf 0xb gesetzt werden.

ReinhardKeil, 2025-06-10:
@FerdinandEn thanks for posting the solution.  The team will add sensible default values for cache handling in the next days.

I like also your proposal to make such values configurable. See my feature request for CMSIS-Toolbox.

jkrech, 2025-07-11:
The use of Arm Debugger has been documented here: https://mdk-packs.github.io/vscode-cmsis-solution-docs/debug.html#arm-debugger