ARM-software/vscode-cmsis-csolution #258: control auto update of tasks.json

ReinhardKeil, 2025-06-03:
For `launch.json` the "cmsis" node controls the automatic update of the file.

A similar feature should be added to tasks.json.

Possible example:

```json
        {
            "label": "CMSIS Load",
            "type": "shell",
            "command": [
                "echo Connect > .cmd.jlink &&",
                "echo LoadFile \"out/Blinky/AppKit-E7/Debug/Blinky.hex\" >> .cmd.jlink &&",
                "echo Exit >> .cmd.jlink &&",
                "jlink -device AE722F80F55D5LS_M55_HP -if SWD -speed auto -commandfile .cmd.jlink"
            ],
            "options": {
                "cwd": "${workspaceFolder}/M55_HE/Blinky"
            },
            "problemMatcher": []
            "cmsis": {
                "updateConfiguration": "auto"
            },
        },
```

jkrech, 2025-07-23:
@JonatanAntoni - please confirm that we will need to introduce a new type instead of using the pre-defined 

`"type": "shell"`

for adding the `"cmsis"` node and the `"updateConfiguration"` node or is there a better approach?
E.g. appending a predefined string to the label that will be ignored when the commands search for the labels?

JonatanAntoni, 2025-07-28:
In order to add further properties to a task we'd probably need to define (and implement) our own task type.

What about renaming the task, e.g. `CMSIS Load` to `CMSIS Load (manual)`? On update the task would only be changed if the name is unchanged. The buttons could just look for a task which `.startsWith('CMSIS Load')` so that any manual suffix is ignored.

The cumbersome part comes when tasks reference other tasks (such as `CMSIS Load+Run`) or launch configs reference a `preLaunchTask`. Renaming a references task requires all references to be updated as well.

jkrech, 2025-08-13:
consider superseding by #307.

jkrech, 2025-10-07:
Especially when using multiple `target-types` or multiple `*.csolution.yml` in a single workspace, preventing the update of the task.json file generation will in most cases lead to failures.
We need to consider an alternative approach, where e.g. the `solution` and `target-set` are stored as part of the task.json in case they do not get automatically updated.

closed 2025-10-29 by ReinhardKeil