ARM-software/vscode-cmsis-csolution #273: Review `cmsis-csolution.actions` in settings.json

ReinhardKeil, 2025-07-01:
Type: Feature Request

There is still a feature that relates to the Arm Debugger and creates a message "Arm Debugger Not Installed".
This should be reviewed an adapted to the new template engine.

[image]

JonatanAntoni, 2025-07-01:
The old functionality is still there but just hidden. Its due to be removed when redesigning the "Manage Solution" dialog. The settings should not be used anymore. Removing them entirely would turn the deprecation message into an unspecific "Unknown Configuration Setting" message.

jkrech, 2025-07-29:
Eve has not completed the review of example repositories yet.