ARM-software/vscode-cmsis-csolution #276: Correlate outdated config file messages with components

ReinhardKeil, 2025-07-03:
From a user point of view it is hard to find the outdated config files in the CMSIS View.  The report in the status bar refers files, but not components.

[image]

jkrech, 2025-07-04:
@brondani, I guess this should be addressed in the CMSIS-Toolbox rather than the extension.

brondani, 2025-07-04:
> I guess this should be addressed in the CMSIS-Toolbox rather than the extension.

The warning messages could be easily extended with the component identifier, but this would increase the verbosity.

Maybe a cleaner solution could be to add icons next to such files/components.
This information is already present and classified in cbuild.yml, for example:
``` yml
   components:
    - component: ARM::Device:Startup&C Startup@2.0.0
      ...
      files:
        - file: ../../../../RTE/Device/SSE-300-MPS3/startup_SSE300MPS3.c
          category: source
          attr: config
          version: 1.1.3
          base: ../../../../RTE/Device/SSE-300-MPS3/startup_SSE300MPS3.c.base@1.1.1
          update: ../../../../RTE/Device/SSE-300-MPS3/startup_SSE300MPS3.c.update@1.1.3
          status: update suggested
```

jkrech, 2026-03-18:
Addressed by #103