ARM-software/vscode-cmsis-csolution #278: Example project load does not respect `target-set:` `images:`

ReinhardKeil, 2025-07-03:
Multi-core examples specify a target-set as shown below.

```yml
 target-types:
    - type: Alif-AppKit-E7
      board: Alif Semiconductor::AppKit-E7-AIML
      device: Alif Semiconductor::AE722F80F55D5LS
      target-set:
        - set:
          images:
            - project-context: M55_HP.Debug
            - project-context: M55_HE.Debug
```

However, when such an example is loaded without a `*.cbuild-set.yml` file, only the first project is selected. Hence the above configuration is ignored.

[image]

ReinhardKeil, 2025-07-07:
Will be implemented with #235

jkrech, 2025-10-15:
```
  target-types:
    - type: STM32H745ZGTx
      device: STM32H745ZGTx
      target-set:
        - set: # List of different build configurations.
          images:
            - project-context: CM4.Debug
            - project-context: CM7.Debug
          debugger:
            name: ST-Link@pyOCD
            port: 3333
            protocol: swd
            clock: 5000000
```

[image]

closed 2025-10-15 by jkrech