ARM-software/vscode-cmsis-csolution #297: RTX5 EventRecorder unavailable

flair0626, 2025-07-23:
Type: Bug Report

**Describe the bug**
OS and Version: WIN10
VS Code Version: 1.102
Extension Version:1.20
KEIL Version:5.42
Target Device:GD32E513

**Screenshots**
With the same configuration and code, creating a project in Keil MDK5 can display RTX event information normally:
[image]

The same configuration and code result in the RTX event information not being displayed when creating a project in Keil MDK6：
[image]

jkrech, 2025-07-24:
@flair0626, how did you resolve the above reported problem?

flair0626, 2025-07-25:
> [@flair0626](https://github.com/flair0626), how did you resolve the above reported problem?

Select source code rather than library in component management.

jkrech, 2025-07-25:
RTX Library variant does not generate events. 
The source variant needs to be selected for RTX as documented in [RTX docs](https://arm-software.github.io/CMSIS-RTX/latest/cre_rtx_proj.html#cre_rtx_proj_er).