ARM-software/vscode-cmsis-csolution #320: Error: there is no registered task type 'virtual-hardware.run'

ReinhardKeil, 2025-09-05:
Bug Report

**Describe the bug**
- Extension Version: 1.60.1-9-g25682b1

I'm getting this message 

```
Error: there is no registered task type 'virtual-hardware.run'. Did you miss installing an extension that provides a corresponding task provider?
Activating task providers all
Activating task providers all
Activating task providers all
```

**To Reproduce**

1. Use https://github.com/Arm-Examples/AVH-Hello
2. Open csolution.yml file
3. Output - Tasks

**Screenshots**

[image]

jreineckearm, 2025-09-05:
The Arm Virtual Hardware Extension was removed from Keil Studio Pack as requested for v1.20.0.
We apparently missed this use case.

Please work around by manually installing https://marketplace.visualstudio.com/items?itemName=Arm.virtual-hardware

Let's discuss if to
- document the Arm Virtual Hardware extension as required for the SDS use case.
- bring it back as is.
- move the task definition or comparable elsewhere (e.g. [debug adapter registry](https://github.com/Open-CMSIS-Pack/debug-adapter-registry/blob/main/templates/FVP.adapter.json))

jreineckearm, 2025-09-05:
Having written that, I am surprised it even flags this. I can't see a `tasks.json` stored with the example on a first glance. Let me double-check.