ARM-software/vscode-cmsis-csolution #326: Create new solution flow broken for target devices

KeilChris, 2025-09-15:
Type: Bug Report

- OS and Version: MacOS 15.6.1/Windows 11 ARM64
- VS Code Version: 1.104.0
- Extension Version: 1.60.0
- Target Device: Any (for example, select the first from the list, A31G111)

The "create new solution" dialog does not filter "Templates, Reference Applications, and Examples" when starting with a **Target Device**. It simply shows the full list of example projects that are installed locally.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to CMSIS view, select "Create New Solution".
2. Click on Target Device dorp-down and select the first device **A31G111**.
3. Expand "Templates, Reference Applications, and Examples".
4. All locally installed examples are shown iwthout any filter

**Expected behavior**
The examples list should be filtered for the selected target device (if this is part of an example for a board). If there's no filtered example available, at least the "Blank Solution" should be shown (which is currently not even available).

**Screenshots**
[image]

brondani, 2025-09-15:
A `device` filter parameter means "filter for compatible boards where this device is mounted".
However the absence of such compatible boards was being wrongly treated as the absence of the filter parameters which is not the case.
This is going to be fixed in https://github.com/Arm-Debug/devtools-external/pull/1315

brondani, 2025-09-16:
PR was merged, it will be available in the next toolbox nightly.