ARM-software/vscode-cmsis-csolution #329: Build Process should use Ninja overwrite message features

ReinhardKeil, 2025-09-16:
Type: Feature Request

Ninja by default prints one line per finished build step and overwrites it in-place if the command succeeds without warnings or errors.

Example:

```txt
[1/151] Generating include/generated/version.h
-- Zephyr version: 3.6.0 (C:/w/cmsis-zephyr/Alif_E7_Zephyr/ipm_arm_mhuv2/zephyr), build: zas-v1.2-177-ga92ae0ad42df
[61/151] Building C object rtss_he/zephyr/lib/libc/minimal/CMakeFiles/lib__libc__minimal.dir/source/string/strstr.c.obj
```

This requires the following changes:

- On Windows, use Powershell as the default terminal console for build
- Build one project at a time (one BUILD_COMMAND)

```
 # Remove the extra echo in cmake scripts
 # BUILD_COMMAND       ${CMAKE_COMMAND} -E echo "Building CMake target '${CONTEXT}'"
```

This implies that project interdependencies are handled by the csolution rpc system and no longer by CMake itself.

brondani, 2025-11-04:
> Build one project at a time (one BUILD_COMMAND)

Note that this breaks interdependencies among build contexts, execute notes and now also zephyr/west build contexts. The build orchestration would need to be significantly rewritten and the inter-context dependency handling currently solved by cmake/ninja would need to be re-implemented (reinventing the wheel?).

Also note that `csolution rpc` does not touch the build, it runs only up to the convert solution and does not call any other tool.
The build is orchestrated by `cbuild` by calling `cbuild2cmake`, `cmake`, etc.

ReinhardKeil, 2025-11-04:
@brondani there are other related issues too, such as https://github.com/Open-CMSIS-Pack/devtools/issues/2210.
So I believe there are also other cases that have issues with the current inter-context handling